zauberzeug/lizard #5: difficulty running a stepper motor

ipsod, 2023-02-11:
I haven't actually plugged a motor in, I'm just playing through the monitor.py script

Output of session below.  Notice the end, where it claims to have non-zero speed, but position stays the same.  I had many instances of stepper state freezing (once at speed 0, I think), and this is just the one I captured and shared.

> 
> Connecting to /dev/ttyUSB0
> > motor = StepperMotor(12, 13)
> I (10974) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| I
> ntr:0 
> I (10974) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| I
> ntr:0 
> > motor.position
> -1
> > motor.position(0, 500, 500)
> > motor.position
> -1
> > motor.speed(1, 500, 500)
> error processing uart0: unexpected number of arguments
> > motor.speed(500, 500)
> > motor.speed
> 500
> > motor.position
> 2283
> > motor.position(0, 500, 500)
> > motor.position
> 7368
> > motor.position
> 6968
> > motor.position
> 6488
> > motor.position
> 6013
> > motor.position
> 5548
> > motor.position
> 5093
> > motor.position
> 4663
> > motor.position
> 4228
> > motor.speed
> -500
> > motor.speed
> -500
> > motor.position
> 753
> > motor.position
> 238
> > motor.position
> 238
> > motor.position
> 238
> > motor.position
> 238
> > motor.speed
> -488
> > motor.position
> 238
> > motor.speed
> -488
> >

This could be related to an error I had when flashing:

> ➜  lizard_firmware_and_devtools_v0.1.4 sudo ./flash.py /dev/ttyUSB0
> [sudo] password for self:          
> Initializing ESP...
> Configuring pins...
> echo 216 > /sys/class/gpio/export
> could not write 216 to export
> echo  50 > /sys/class/gpio/export
> could not write 50 to export
> echo out > /sys/class/gpio/gpio216/direction
> could not write out to gpio216/direction
> echo out > /sys/class/gpio/gpio50/direction
> could not write out to gpio50/direction
> Bringing microcontroller into flash mode...
> echo   0 > /sys/class/gpio/gpio216/value
> could not write 0 to gpio216/value
> echo   0 > /sys/class/gpio/gpio50/value
> could not write 0 to gpio50/value
> echo   1 > /sys/class/gpio/gpio216/value
> could not write 1 to gpio216/value
> Flashing...
> esptool.py v3.1
> Serial port /dev/ttyUSB0
> Connecting.......
> Chip is ESP32-D0WDQ6 (revision 1)
> Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
> Crystal is 40MHz
> MAC: 24:62:ab:b0:87:34
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 921600
> Changed.
> Configuring flash size...
> Auto-detected Flash size: 4MB
> Flash will be erased from 0x00001000 to 0x00006fff...
> Flash will be erased from 0x00008000 to 0x00008fff...
> Flash will be erased from 0x00010000 to 0x0011afff...
> Flash params set to 0x0220
> Compressed 24416 bytes to 15713...
> Wrote 24416 bytes (15713 compressed) at 0x00001000 in 0.5 seconds (effective 397.2 kbit/s)
> ...
> Hash of data verified.
> Compressed 3072 bytes to 105...
> Wrote 3072 bytes (105 compressed) at 0x00008000 in 0.0 seconds (effective 531.4 kbit/s)...
> Hash of data verified.
> Compressed 1092368 bytes to 573275...
> Wrote 1092368 bytes (573275 compressed) at 0x00010000 in 9.0 seconds (effective 975.7 kbit
> /s)...
> Hash of data verified.
> 
> Leaving...
> Hard resetting via RTS pin...
> Bringing microcontroller into normal operation mode...
> echo   1 > /sys/class/gpio/gpio50/value
> could not write 1 to gpio50/value
> echo   0 > /sys/class/gpio/gpio216/value
> could not write 0 to gpio216/value
> echo   1 > /sys/class/gpio/gpio216/value
> could not write 1 to gpio216/value
> echo 216 > /sys/class/gpio/unexport
> could not write 216 to unexport
> echo  50 > /sys/class/gpio/unexport
> could not write 50 to unexport

Side note, there was no esptool.py included in the zip release file, though the instructions referred to it.

falkoschindler, 2023-02-15:
Thanks for reaching out, @ipsod!

You are correct: The speed has not been updated after reaching the target position. This should be fixed in [88c5a18](https://github.com/zauberzeug/lizard/commit/88c5a182afedc975d05de2b7515598f24d0f9d16).

The output mentioning /sys/class/gpio has nothing to do with it. The flash script could certainly be improved in this regard. Toggling GPIO pins is irrelevant if the ESP is connect via USB.

closed 2023-02-16 by falkoschindler

rodja, 2023-02-17:
We have just released [v0.1.5](https://github.com/zauberzeug/lizard/releases/tag/v0.1.5) containing this fix.