zauberzeug/lizard #24: core.print breaks some characters

falkoschindler, 2024-01-10:
Minimum reproduction:
```py
core.print('boo\n')
```

This doesn't print "boo", but only "oo". The "b" is converted to a backspace.

- The following characters are affected: "b", "f", "n", "r" and "t".
- The problem doesn't occur without the trailing newline "\n".

Somehow these five characters are converted into their escaped counterparts backspace, form feed, new line, carriage return and tab.

JensOgorek, 2024-12-02:
Adding more characters after the `\n` gets even funnier:
```
> core.print("boo\nbooboo")
oo
ooo"
```
```
> core.print("boo\nboobooboo")
oo
oooo"
```
```
> core.print("boo\nbooboobiibaa")
oo
ooiaa"
```
```
> core.print("boo\nbooboobiibaauu")
oo
ooiaauu"
```
```
> core.print("boo\nbooboobiibaauub")
oo
ooiaau"
```
There is something funny happening in the parser, I guess.

JensOgorek, 2024-12-02:
It is definitely a problem how the parser handles ESCAPE CHARs
`#define ESCAPE_CHAR(c, info) ((c) == 'b' ? '\b' : (c) == 'f' ? '\f' : (c) == 'n' ? '\n' : (c) == 'r' ? '\r' : (c) == 't' ? '\t' : (c))`
in its code.

falkoschindler, 2024-12-02:
Maybe we can reproduce it online?
https://ianh.github.io/owl/try/

JensOgorek, 2024-12-05:
When processing escape sequences (like '\n'), the code used the original string length instead of the processed length. For example, "\n" has length 2, but should be length 1 after processing.

closed 2024-12-05 by falkoschindler