zauberzeug/lizard #53: Expander cannot be flashed

falkoschindler, 2024-07-09:
Currently there is an issue flashing the expander, causing it to crash with a backtrace. We assume the problem is due to a different partition table in version 0.1.12. So when updating an expander from some version like 0.1.5 to 0.1.11, it works well. But upgrading to 0.1.12 fails.

Maybe we can make `CONFIG_PARTITION_TABLE_OFFSET` an optional parameter for `core.flash()` to work around these edge cases. Or we describe how to OTA-update the expander in the readme. As long as we don't change the partition table in the near future, a temporary fix might be good enough.

falkoschindler, 2024-07-10:
> Or we describe how to OTA-update the expander in the readme. 

We noticed that this is not an option, because the firmware on the expander doesn't know the `ota()` command. So we need to fix the `flash()` command to work with arbitrary partition tables on the target ESP.

JensOgorek, 2024-07-12:
Fixed with PR: #56

closed 2024-07-15 by falkoschindler