zauberzeug/lizard #57: Bootloop

Johannes-Thiel, 2024-07-17:
When the following script is configured, the esp will Bootloop.
```
!+s1 = Serial(26,27,115200,1)
!+s2 = Serial(13,4,115200,2)
!+can = Can(32,33,1000000)
```

When the can is initialized before the Serial, the esp will not Bootloop.
```
!+can = Can(32,33,1000000)
!+s1 = Serial(26,27,115200,1)
!+s2 = Serial(13,4,115200,2)
```

Before the the Bootloop follwing error is thrown:
```
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x40093110
file: "IDF/components/driver/twai.c" line 465
func: twai_driver_install
expression: esp_intr_alloc(ETS_TWAI_INTR_SOURCE, g_config->intr_flags, twai_intr_handler_main, NULL, &p_twai_obj->isr_handle)
abort() was called at PC 0x40093113 on core 0
```

falkoschindler, 2024-07-17:
@Johannes-Thiel Do you, by any chance, see a stack trace when abort() is called?

Johannes-Thiel, 2024-11-15:
I tested this issue again. Our update tho ESP IDF 5.3.1 has fixed this issue. The ESP does not bootloop with both configs.

closed 2024-11-15 by Johannes-Thiel