zauberzeug/lizard #65: "invalid token" errors when using steering with Bluetooth

rodja, 2024-08-26:
It seems that https://github.com/zauberzeug/field_friend/issues/147 is caused by some kind of issue in Lizard. When I send steering commands I can see that the App successfully sends a `wheels.speed(0.0000,0.0000)` command, but the wheels keep on spinning: 

https://github.com/user-attachments/assets/a988c1cb-9b52-4a30-9d29-27fed7c300a2

I noticed that there are numerous error messages `invalid token at range 0 1 "w"@74` appearing in the serial log while steering. They do not show if the robot is idle and the app does not send anything.

Johannes-Thiel, 2024-08-26:
it could be that the robot stops because of this line in the startup of the robot.
`when core.last_message_age > 1000 then wheels.speed(0, 0); end` 
This will stop the robot, if the core does not get a command in 1000 milli seconds.

rodja, 2024-09-11:
Yes @JensOgorek. But why? The full text is send via mobile to Bluetooth. Is the error in the lizards bluetooth code? Or in the parsing?

JensOgorek, 2024-09-11:
Is the app directly sending the commands to the Lizard Bluetooth api? If yes, I would say it's most likely with the Bluetooth itself. The message handler is using pattern detection on driver level to look for '\n' in the messages buffer. It did not overflow, since it would throw at least a warning. 
So I think it might be the message itself in the Bluetooth code or from the app itself. If the message is correct until then, we might have to dive deeper and see what's wrong.

JensOgorek, 2024-09-26:
We could not reproduce this with ESP32 or a Robot Brain. Maybe @rodja check if you can break it again with idf 5.3.1 update

rodja, 2024-11-29:
Thanks to #107, this issue can be closed.

closed 2024-11-29 by rodja