zauberzeug/lizard #66: Unknown Property "level"

Johannes-Thiel, 2024-09-05:
there seems to be a issue regarding the level property. This has been observed on the mini robot z33 and on the f15.
```py
 read: I (2450) gpio: GPIO[34]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
2024-09-05 11:41:52.975 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: I (2450) gpio: GPIO[35]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
2024-09-05 11:41:52.985 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: I (2470) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
2024-09-05 11:41:52.995 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: error while loading startup script: unknown property "level"@08
```
Due to this error, there is no core output.

falkoschindler, 2024-09-05:
We noticed a similar problem. I suspect you're using a `level` property on a port expander inside a rule. PR #51 should have fixed the problem with properties being evaluated too early, so we removed the workaround for `level` and `active`. But it turned out that when compiling a property expression, the property is still accessed in order to retrieve its data type.

A workaround (and maybe the only valid solution) is to add properties explicitly before defining rules:

```py
button = p0.Input(42)
button.level = 0
when button.level == 1 then do_something(); end
```

falkoschindler, 2024-09-05:
This issue will be solved by #18:
"When instantiating a proxy module, the execution should wait until the expander has sent a first broadcast message."

JensOgorek, 2024-11-22:
Fixed with #104

closed 2024-11-22 by JensOgorek