zauberzeug/lizard #75: Trouble flashing p0

SeaTechRC, 2024-10-14:
Hello,

I have trouble flashing the p0 expander on the robot brain.
See the log:
[image]

To minimize interference of other components, I have tried writing the following startup configuration manually:
```
!-
!+serial = Serial(26, 27, 115200, 1)
!+p0 = Expander(serial, 25, 14)
!.
```
and then simply `p0.flash()`
I fear is, that this might actually be a flash page that is broken.

SeaTechRC, 2024-10-14:
[https://github.com/espressif/esp-serial-flasher/blob/a4f5033895489be0f1f75d254d3a040a9231560b/private_include/protocol.h#L79](The comment) in the header file actually explains it I think. In that case it should be a broken flash page.

falkoschindler, 2024-10-17:
As it turned out, some strapping pins have not been in the correct state.
We will add a check before flashing so that the user sees an informative error message.