zauberzeug/lizard #87: Serial read_line behavior with multiple lines in RX buffer

SeaTechRC, 2024-11-05:
Hello,

As mentioned in #78, if I am not misunderstanding the ESP-IDF uart_driver, calling Serial::read_line() when 2 lines are in the RX buffer causes the pattern position for the second line to be incorrect.
As far as I could see, the pattern positions are not updated when the RX buffer in the uart buffer gets partly read.

As Example:
The pattern is "\n"
The uart RX buffer was empty (aswell as the pattern queue) and was filled with: 
`test 1\ntest2\n`
without any reads inbetween.
The driver would have the following pattern positions in queue:
`6, 12`

When Serial::read_line() is called, it retrieves the data up to the first pattern in the queue. After read_line, the buffer and pattern pos queue should be:
`test 2\n`
`12`

If Serial::read_line() gets called again, we retrieve parts of the next message that gets put into the buffer.

Do note:
I have not yet been able to confirm that this is actually the case.

Kind regards

JensOgorek, 2024-12-05:
We will take a look at this soon

SeaTechRC, 2025-05-20:
Upon taking a second look, the function [uart_read_bytes](https://github.com/espressif/esp-idf/blob/c8fc5f643b7a7b0d3b182d3df610844e3dc9bd74/components/esp_driver_uart/src/uart.c#L1517) updates the pattern positions by use of uart_pattern_queue_update.

closed 2025-05-20 by SeaTechRC