zauberzeug/lizard #88: IMU not working

pascalzauberzeug, 2024-11-05:
Setting up an IMU is not working. We tested 0.5.2, but I guess the bug was introduced with the IDF Update in 0.5.0. It works with 0.4.0

```
2024-11-05 00:17:43.994 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2259) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.004 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2259) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.014 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2259) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.025 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2259) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.035 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2259) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.055 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2259) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.065 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2269) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.076 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2269) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.086 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2269) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.097 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2279) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.108 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2279) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.118 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2279) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.129 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2289) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.139 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2289) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.150 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2289) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.166 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2299) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.177 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2299) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.187 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2299) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.198 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2309) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.209 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2309) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.219 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2309) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.230 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2319) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.240 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2319) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.251 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2319) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.261 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2329) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.278 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2329) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.289 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2329) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.299 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2339) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.310 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2339) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.320 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2339) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.331 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2349) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.342 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2349) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.352 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2349) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.363 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2359) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.373 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2359) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.390 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2359) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.400 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2369) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.411 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2369) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.422 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2369) BNO055: (i2c WL) W (2369) uart: Fail to enqueue pattern position, pattern queue is full.
2024-11-05 00:17:44.432 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: Error: 263W (2379) uart: Fail to enqueue pattern position, pattern queue is full.
2024-11-05 00:17:44.443 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read:
2024-11-05 00:17:44.454 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2389)W (2389) uart: Fail to enqueue pattern position, pattern queue is full.
2024-11-05 00:17:44.464 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read:  BNO055: (W (2399) uart: Fail to enqueue pattern position, pattern queue is full.
2024-11-05 00:17:44.475 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: i2c WL) ErrW (2399) uart: Fail to enqueue pattern position, pattern queue is full.
2024-11-05 00:17:44.485 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: or: 263
2024-11-05 00:17:44.502 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2409) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.513 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2409) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.524 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2419) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.534 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2419) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.545 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2419) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.555 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2429) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.566 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2429) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.576 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2429) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.587 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2439) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.597 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2439) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.613 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2439) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.624 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2449) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.635 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2449) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.645 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2449) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.656 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2459) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.666 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2459) BNO055: (i2c WL) Error: 263
2024-11-05 00:17:44.677 [DEBUG] rosys/hardware/communication/serial_communication.py:72: read: E (2459) BNO055: (i2c WL) Error: 263
[... cut ...]
# Socket 5 iigiir_v02
s5_in_1 = p0.Input(35)
s5_in_2 = p0.Input(18)
s5_in_3 = p0.Input(21)
s5_in_4 = p0.Input(19)
str socket_5 = "iigiir_v02"

# Socket 6 empty

imu = Imu()

rdyp_status = Input(39)
vdp_status = p0.Input(39)

core.output("core.millis core.heap rdyp_status.level vdp_status.level socket_1 s1_in_1.level socket_2 s2_in_1.level s2_in_2.level socket_3 s3_in_1.level socket_5 s5_in_1.level s5_in_2.level s5_in_3.level s5_in_4.level")

rdyp = Output(15)
en3 = Output(12)
en3.on()
rdyp.on()
```

JensOgorek, 2024-11-05:
The IMU module itself did not change. But the BNO one did. 
The error is coming from [if (id != 0xA0) throw BNO055ChipNotDetected();](https://github.com/zauberzeug/lizard/blob/main/main/modules/BNO055ESP32.cpp#L803). That is related to a change made for idf 5.3.1 with         `esp_rom_gpio_pad_select_gpio(_intPin);`

JensOgorek, 2024-11-05:
Yet, almost nothing was changed with the IMU and the BNO module.

pascalzauberzeug, 2024-11-05:
When we last spoke, @Johannes-Thiel had a fix but wasn't quite sure yet.

closed 2024-11-14 by Johannes-Thiel