zauberzeug/lizard #100: Proxy message optimization

JensOgorek, 2024-11-18:
When multiple proxies get created for the same expander module, each will create their own broadcast. To reduce overhead and unnecessary spam, there should only be one broadcast message.

falkoschindler, 2024-11-19:
@JensOgorek Why do you think multiple broadcast messages produce "overhead and unnecessary spam"? As far as I understand, no information is sent multiple times.

Sure, combining multiple messages into a single broadcast might eliminate line breaks and the number of checksums. But one could argue that sending the broadcast in parts is more robust, since a single transmission error wouldn't invalidate the whole broadcast.

JensOgorek, 2024-11-28:
With the optimization of owl in #107 this is not really needed anymore, since we limit the size of owl directly.

closed 2024-11-28 by JensOgorek