zauberzeug/lizard #99: Core gets unresponsive (long startup scripts)

JensOgorek, 2024-11-18:
When a startup script is too long, the core gets unresponsive.

JensOgorek, 2024-11-18:
Also, this issue would be hidden by #97, since the hotfix could be erased from the startup scripts

falkoschindler, 2024-11-19:
@JensOgorek Do we have a reproduction?

Johannes-Thiel, 2024-11-20:
The code breaks at the line `auto const tree = std::unique_ptr<owl_tree, std::function<void(owl_tree*)>>(owl_tree_create_from_string(line), owl_tree_destroy);` (l.319 in main.cpp). Until this line the Programm runs as expected.
When the core.debug is true the debug messeges are 
 ```
vdp_status.active=false;vdp_status.change=-1;vdp_status.inverted=false;vdp_status.level=0;
Tree creation took 0.274 ms
error: invalid token at range 0 1 "v"
```

When the code is running as expected the debug will be:
```
vdp_status.active=true;vdp_status.change=0;vdp_status.inverted=false;vdp_status.level=1;
Tree creation took 1.942 ms
statements (0 - 88)
  statement (0 - 22)
    property_assignment (0 - 22)
      identifier@module_name - vdp_status (0 - 10)
      identifier@property_name - active (11 - 17)
      expression : TRUE (18 - 22)
  statement (23 - 42)
    property_assignment (23 - 42)
      identifier@module_name - vdp_status (23 - 33)
      identifier@property_name - change (34 - 40)
      expression : INTEGER (41 - 42)
        integer - 0 (41 - 42)
  statement (43 - 68)
    property_assignment (43 - 68)
      identifier@module_name - vdp_status (43 - 53)
      identifier@property_name - inverted (54 - 62)
      expression : FALSE (63 - 68)
  statement (69 - 87)
    property_assignment (69 - 87)
      identifier@module_name - vdp_status (69 - 79)
      identifier@property_name - level (80 - 85)
      expression : INTEGER (86 - 87)
        integer - 1 (86 - 87)
  statement (88 - 88)
    noop (88 - 88)
Tree traversal took 0.991 ms
```
It looks like the problem is in the owl tree cration. 
When the string of the line is printed it is complete. This error ocuurs only for the messages from the expander, that are send via serial and handelt by the message_handler.

Johannes-Thiel, 2024-11-20:
When comparing the string that is to be used for the owl_tree  there is no difference between the working and the defective function call.
Defective:
```
String hex dump: 63 6F 72 65 2E 6D 75 74 65 28 29
>> core.mute()
Tree creation took 0.358 ms
error: invalid token at range 0 1 "c"
```
Working:
```
String hex dump: 63 6F 72 65 2E 6D 75 74 65 28 29
>> core.mute()
Tree creation took 1.563 ms
statements (0 - 11)
  statement (0 - 11)
    method_call (0 - 11)
      identifier@module_name - core (0 - 4)
      identifier@method_name - mute (5 - 9)
Tree traversal took 0.699 ms
```
As can be seen with the Hex output the stings are the same.

rodja, 2024-11-22:
I just heard from @pascalzauberzeug that the long startup scripts are working with version 5.2. With a minimal reproduction it should be fairly simple to find the code which breaks the newer version with git bisect.

JensOgorek, 2024-11-22:
> I just heard from @pascalzauberzeug that the long startup scripts are working with version 5.2. With a minimal reproduction it should be fairly simple to find the code which breaks the newer version with git bisect.

I checked multiple versions of Lizard with the startup script I got from @Johannes-Thiel.
I configured the core esp with two versions of that script, one with a "core.print("....[lot of dots]...core.millis")" line and one without.

Core Esp flashed with -e flag.
p0 created and flashed with core esp.

1. 0.5.2 bin from release
1.1 normal script
- script gets transferred
- core is responsive
- script is readable with !?
- p0 does exist
- p0 is configured and sends data

1.2 After adding: core.print("....[lot of dots]...core.millis")
- error: invalid token at range 0 1 "v" and so on
- core unresponsive

2. Version from PR #78 (with added casting fix in serial.cpp)
2.1 normal script
- script gets transferred
- core is responsive
- script is readable with !?
- p0 does exist
- p0 is configured and sends data

2.2 After adding: core.print("....[lot of dots]...core.millis")
- error: invalid token at range 0 1 "f" and so on
- core unresponsive

3. Main branch version before last PR, so basically PR #98 with expander improvements
3.1 normal script:
- script gets transferred
- core is **unresponsive**
- script is readable with !?
- p0 does exist
- p0 is configured and sends data

3.2 After adding: core.print("....[lot of dots]...core.millis")
- same as 3.1

The bug happens in other versions as well, but faster in the newer versions with expander improvements.
The invalid tokens are indeed the messages coming from the p0, but it will also display core messages the same way.

Johannes-Thiel, 2024-11-25:
<details><summary>I found a script, that is breaking. This breaks without any P0 and any serial.</summary>

```
rdyp = Output(15)
en3 = Output(12)
can = Can(32, 33, 1000000)
bluetooth = Bluetooth("fieldfriend-f13")
l0 = ODriveMotor(can, 0)
r0 = ODriveMotor(can, 512)
l1 = ODriveMotor(can, 256)
r1 = ODriveMotor(can, 768)
l0.m_per_tick = 0.03953674121405751
r0.m_per_tick = 0.03953674121405751
l1.m_per_tick = 0.03953674121405751
r1.m_per_tick = 0.03953674121405751
l0.reversed = false
r0.reversed = true
l1.reversed = false
r1.reversed = true
wheels = ODriveWheels(l0, r0)
wheels_front = ODriveWheels(l1, r1)
wheels.width = 0.47
wheels_front.width = 0.47

yaxis_motor = CanOpenMotor(can, 96)
yaxis_end_l = Input(13)
yaxis_end_l.inverted = true
yaxis_end_r = Input(4)
yaxis_end_r.inverted = true
yaxis_end_l.level = 0
yaxis_end_l.active = false
yaxis_end_r.level = 0
yaxis_end_r.active = false
yaxis = MotorAxis(yaxis_motor, yaxis_end_r, yaxis_end_l)

tornado_motor_z = ODriveMotor(can, 1280)
tornado_motor_turn = ODriveMotor(can, 1024)
tornado_motor_z.m_per_tick = 0.0019968051118210866
tornado_motor_turn.m_per_tick = 0.07987220447284345
tornado_motor_z.limits(1.5, 30)
tornado_motor_turn.limits(1.5, 30)
tornado_motor_z.reversed = true
tornado_motor_turn.reversed = true
tornado_end_top = Input(33)
tornado_end_top.inverted = true
tornado_end_bottom = Input(4)
tornado_end_bottom.inverted = true
tornado_ref_motor = Input(5)
tornado_ref_motor.inverted = true
tornado_ref_gear = Input(36)
tornado_ref_gear.inverted = false
tornado_ref_knife_stop = Input(5)
tornado_ref_knife_stop.inverted = false
tornado_ref_knife_ground = Input(32)
tornado_ref_knife_ground.inverted = true

\# TODO: remove when lizard issue 66 is fixed.
tornado_end_top.level = 0
tornado_end_top.active = false
tornado_end_bottom.level = 0
tornado_end_bottom.active = false
tornado_ref_motor.level = 0
tornado_ref_motor.active = false
tornado_ref_gear.level = 0
tornado_ref_gear.active = false
tornado_ref_knife_stop.level = 0
tornado_ref_knife_stop.active = false
tornado_ref_knife_ground.level = 0
tornado_ref_knife_ground.active = false

[... 123 lines cut ...]
core.print("start")
core.print("start")
core.print("start")
core.print("start")
core.print("start")
core.print("start")
core.print("start")
core.print("start")
core.print("start")
core.print("start")
core.print("start")
core.print("start")
core.print("start")
core.print("start")
core.print("start")
core.print("start")

core.print("........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................core.millis")
core.output("core.millis")
rdyp.on()
en3.on()
core.print("done")
```
</details> 
This shows that the Problem does not come from the P0 or serial.

JensOgorek, 2024-11-25:
I was analyzing the parser.h today and went through the complete calls.
For example, a Tree with 4 properties from the p0 expander looks like this:
Incoming message:

bumper_front_bottom.active=true;bumper_front_bottom.change=0;bumper_front_bottom.inverted=false;bumper_front_bottom.level=1;

Owl tree:
- **statements (0 - 124)**
  - **statement (0 - 31)**
    - property_assignment (0 - 31)
      - identifier@module_name: bumper_front_bottom (0 - 19)
      - identifier@property_name: active (20 - 26)
      - expression: TRUE (27 - 31)
  - **statement (32 - 60)**
    - property_assignment (32 - 60)
      - identifier@module_name: bumper_front_bottom (32 - 51)
      - identifier@property_name: change (52 - 58)
      - expression: INTEGER (59 - 60)
        - integer: 0 (59 - 60)
  - **statement (61 - 95)**
    - property_assignment (61 - 95)
      - identifier@module_name: bumper_front_bottom (61 - 80)
      - identifier@property_name: inverted (81 - 89)
      - expression: FALSE (90 - 95)
  - **statement (96 - 123)**
    - property_assignment (96 - 123)
      - identifier@module_name: bumper_front_bottom (96 - 115)
      - identifier@property_name: level (116 - 121)
      - expression: INTEGER (122 - 123)
        - integer: 1 (122 - 123)
  - **statement (124 - 124)**
    - noop (124 - 124)
------------------------------------

Before a tree gets created, the owl parser will tokenize the incoming string. After some digging, I noticed the tokens are not generated at all for the faulty input.
Long story short: The `struct owl_token_run *run = malloc(sizeof(struct owl_token_run));` in the [`owl_default_tokenizer_advance` ](https://github.com/zauberzeug/lizard/blob/main/main/parser.h#L1845) function is failing. We already checked the memory, and it is not overflowing, but there is something else failing.
Now we know where the problem is created.
Next I will try to find out what causes this to fail and how to fix it.

rodja, 2024-11-26:
> Long story short: The struct owl_token_run *run = malloc(sizeof(struct owl_token_run)); in the [owl_default_tokenizer_advance ](https://github.com/zauberzeug/lizard/blob/main/main/parser.h#L1845)function is failing

Does this also happen in isolation on your pc? Eg. not running on the ESP?

JensOgorek, 2024-11-26:
> Does this also happen in isolation on your pc? Eg. not running on the ESP?
Not yet, I would try that next. But my guess would be, that it would work without problems. There is something going on within the ram. 
If I revert the `while (n < size)` change we do with the gen_parser to its original state  `while (n < size|| n < 4096)`, the owl tree runs for ~30 seconds, instead of just 4 calls. This directly changes how the memory gets reallocated while growing the tree. My best guess would be memory fragmentation right now, since malloc can only allocate contiguous memory.

JensOgorek, 2024-11-26:
It went quite hidden but yet obvious:
The struct owl is creating for its Token run is enormous.
```
>> Token run size: 40968
>> Available heap size: 40248
>> Enough space in heap: No
```

closed 2024-11-28 by falkoschindler