zauberzeug/lizard #152: Espresso.py Device or resource busy on fail

JensOgorek, 2025-06-17:
After failing to flash the esp32, espresso will block the ESP32 from doing anything. This might be connected to #149 

<details>
  <summary>crashed flash attempt</summary>

```bash
Espresso running...
Flashing...
Configuring EN and G0 pins...
  echo 216 > /sys/class/gpio/export
  echo  50 > /sys/class/gpio/export
  echo out > /sys/class/gpio/gpio216/direction
  echo out > /sys/class/gpio/gpio50/direction
Bringing the microcontroller into flash mode...
  echo   1 > /sys/class/gpio/gpio216/value
  echo   1 > /sys/class/gpio/gpio50/value
  echo   0 > /sys/class/gpio/gpio216/value
  esptool.py --chip esp32 --port /dev/ttyTHS1 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x20000 build/lizard.bin
esptool.py v3.2
Serial port /dev/ttyTHS1
Connecting...
Failed to get PID of a device on /dev/ttyTHS1, using standard reset sequence.
.
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c4:dd:57:8d:83:f0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00020000 to 0x001cefff...
Warning: some reserved header fields have non-zero values. This image may be from a newer esptool.py?
Compressed 25824 bytes to 16288...
Wrote 25824 bytes (16288 compressed) at 0x00001000 in 0.7 seconds (effective 309.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 145...
Wrote 3072 bytes (145 compressed) at 0x00008000 in 0.1 seconds (effective 393.3 kbit/s)...
Hash of data verified.
Compressed 1762000 bytes to 988721...
Writing at 0x0008b604... (19 %)
A fatal error occurred: Serial data stream stopped: Possible serial noise or corruption.
Error: Flashing failed. Use "sudo" and check your parameters.
```

</details>

<details>
  <summary>run afterward</summary>

```bash
Espresso running...
Flashing...
Configuring EN and G0 pins...
  echo 216 > /sys/class/gpio/export
OSError: [Errno 16] Device or resource busy

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/zauberzeug/lizard/./espresso.py", line 209, in <module>
    main(args.command)
  File "/home/zauberzeug/lizard/./espresso.py", line 201, in main
    flash()
  File "/home/zauberzeug/lizard/./espresso.py", line 139, in flash
    with _pin_config():
  File "/home/zauberzeug/.pyenv/versions/3.11.9/lib/python3.11/contextlib.py", line 137, in __enter__
    return next(self.gen)
           ^^^^^^^^^^^^^^
  File "/home/zauberzeug/lizard/./espresso.py", line 55, in _pin_config
    write_gpio('export', EN_PIN)
  File "/home/zauberzeug/lizard/./espresso.py", line 175, in write_gpio
    Path(f'/sys/class/gpio/{path}').write_text(f'{value}\n', encoding='utf-8')
  File "/home/zauberzeug/.pyenv/versions/3.11.9/lib/python3.11/pathlib.py", line 1078, in write_text
    with self.open(mode='w', encoding=encoding, errors=errors, newline=newline) as f:
OSError: [Errno 16] Device or resource busy

```
  
</details>

falkoschindler, 2025-06-22:
@JensOgorek This looks like a duplicate of #149, which should be fixed via PR #151.

JensOgorek, 2025-06-22:
I think you are right, but I will try to break it next time :)

closed 2025-07-17 by JensOgorek